AzonInc/Doorman #1: Produktion

AzonInc, 2024-04-08:
Hier nochmal zusammengefasst:
2x @peteh
1x @calibanorg
1x @alkue1
1x @gee33
1x @AzonInc

peteh, 2024-04-08:
Gibts noch ein groesseres Risiko dass was nicht funktioniert, also dass man ggf 2 Iterationen einplanen muss oder bist du relativ sicher, dass der erste Wurf schon ganz gut passen wird?

AzonInc, 2024-04-08:
Also ich bin relativ zuversichtlich aber das Risiko liegt ja letztenendes bei mir.
Schauen wir mal aber es sollte in der Theorie passen.

alkue1, 2024-04-08:
1x , wenn möglich gerne mit ESP :)

gee33, 2024-04-09:
> @gee33 Willst du dir den ESP32-S3-WROOM-1 selbst besorgen oder soll ich den gleich mit drauf machen?
> 
> Bei der Schraubklemme musst du dann mal schauen - ich verwende folgende im Design: https://wmsc.lcsc.com/wmsc/upload/file/pdf/v2/lcsc/2303171000_Cixi-Kefa-Elec-KF128-2-54-3P_C474921.pdf
> 
> Und sonst sind es ja nur noch 3*1x2P Header sofern du die für GPIO / Relais benötigst.

@AzonInc ja bitte nehm doch den ESP gleich dazu

gee33, 2024-04-09:
ich brauche aber kein case  , ich baue das dann hinter die TC50

AzonInc, 2024-04-10:
> ich brauche aber kein case , ich baue das dann hinter die TC50
Und die Schraubklemme hast du schon verfügbar?

gee33, 2024-04-10:
> > ich brauche aber kein case , ich baue das dann hinter die TC50
> 
> Und die Schraubklemme hast du schon verfügbar? Weil du sagtest du möchtest deine eigenen Teile verwenden.

ja, danke , Schraubklemmen habe ich

dannyyy, 2024-04-19:
Darf man sich einer Sammelbestellung / Produktion noch anschliessen?
Wäre an dem Board inkl ESP32 interessiert. Gerne die SMD Variante wie sie in der README.md abgebildet ist.

Habe bei mir auch ein TC50. Benötige kein Case. Entweder drucke ich es selbst oder finde noch eine andere Lösung.
In welcher Preisregion wird sich das ganze bewegen?

AzonInc, 2024-04-19:
Hi,
ja klar das ist nicht möglich.

Also der endgültige Preis steht noch nicht fest aber wird absolutes Maximum 30€ sein zuzüglich Versand dann.
Hoffen wir mal dass es bei ca. 25 landet. 
Ich prüfe aktuell noch die Verfügbarkeit von Teilen bei verschiedenen Herstellern.

dannyyy, 2024-04-19:
@AzonInc super, dann darfst du mich gerne auf die Liste setzen. Nehme das rundum Wohlfühlpaket. THT Löten kann ich, bei SMD fehlt mir nötige Zubehör (Paste, Heizbett, ...). Du darfst da gerne etwas draufschlagen, für die zusätzliche Arbeit, die allenfalls anfällt. Werde definitiv was dazu sponsern, habe ewig darauf gewartet, dass jemand eine alternative Lösung für Nello / Nuki findet.

zopieux, 2024-05-02:
Hallo!

I hope English is fine? My German isn't great. :sweat_smile: 

I'm also interested in getting the assembled Doorman v1.3 / N8R8, mit Schraubklemme. This is for a two-wire operated TC50. I don't believe I need the relay though. Can you ship to Switzerland, assuming I provide the shipping fee?

Also, I'm a bit confused, is this issue about ordering preassambled boards or PCB + parts + stencil for manual assembly & soldering?

AzonInc, 2024-05-02:
> I hope English is fine? My German isn't great. 😅

Absolutely no problem :)

> I'm also interested in getting the assembled Doorman for a two-wire operated TC50:
> 
> * v1.3 / N8R8
> * with the Schraubklemme
> * without the relay
> * without the enclosure.

Thanks for your interest.
I only ordered 5 of the rev 1.3 boards in order to test them first - that's also the reason why the list continues with rev 1.4.
When everything is fine in rev 1.3 I'll order a few rev 1.4 which only has a minimal change to allow two-wire operation on the bus and gnd wires only.
And as it's full assembly it always includes the relay even if people don't really use it.

I would probably include you in the 1.4 order then if you want :)

> Can you ship to Switzerland, assuming I provide the shipping fee?

Sure that's no problem I guess.

> Also, I'm a bit confused, is this issue about ordering preassambled boards, or rather PCB + parts + stencil for manual assembly & soldering?

It's about ordering the preassembled board.
Revision 1.3 needs manual assembly of the ESP32 (which I would do) but I'm planning to include the ESP32 in general assembly of rev 1.4 as well.
I'm not sure whether I order 5 or more of rev 1.4. Because there is another revision without the ESP32 Module which uses the ESP32-S3 Chip itself. But it doesn't really change the costs in general - just the components footprint.

And I don't include the screw terminal in the PCBA in order to reduce the costs for THT components.
So you can decide which screw terminal you really want to use and I'll solder it for you. But if you want to solder the screw terminal yourself you're free to do it as well haha.

Just let me know how you decide :)

zopieux, 2024-05-02:
Thanks for the quick reply.

>include you in the 1.4 order then if you want

Yes, please count me in! No problem for the screw terminal, if you're doing then tyvm, otherwise I can definitely solder it myself if it can save you time.

smartforcedev, 2024-05-07:
I am in too if this is still open. What are the next steps?

Cool project! Thank you very much!

AzonInc, 2024-05-07:
> I am in too if this is still open. What are the next steps?
> 
> Cool project! Thank you very much!

Glad you like it :)
I ordered 5 REV1.3 boards and will order more as soon as I can verify that they work as designed.
I'll add you to the preorder-list - just tell me your requirements i.e. screw terminal.

dannyduando, 2024-05-08:
Moin,

Ich würde gerne auch noch eins nehmen mit Case ums in die Wand zu schrauben.

AzonInc, 2024-05-08:
@dannyduando auch für TCS dann? und was für eine Schraubklemmen Variante würdest du wollen + in welchem Land wohnst du?

Wlad2288, 2024-05-14:
Hey @AzonInc!
Cooles Projekt ... vielen Dank dafür.
Ich wäre auch interessiert, weiß aber nicht welche Details du von mir benötigst.
Ich habe eine TCS IMM1300 --> https://www.tcsag.de/fileadmin/user_upload/TCS_DE/Metanavigation/Downloads/Produktinformationen/Innenstationen/PI_IMM1300.pdf
Ich hoffe die ist kompatibel.
VG,
Wlad

AzonInc, 2024-05-15:
> Hey @AzonInc!
> 
> Cooles Projekt ... vielen Dank dafür.
> Ich wäre auch interessiert, weiß aber nicht welche Details du von mir benötigst.
> Ich habe eine TCS IMM1300 --> https://www.tcsag.de/fileadmin/user_upload/TCS_DE/Metanavigation/Downloads/Produktinformationen/Innenstationen/PI_IMM1300.pdf
> Ich hoffe die ist kompatibel.
> 
> Ich würde gerne die FW von https://github.com/peteh/doorman flashen und MQTT nutzen.
> 
> VG,
> Wlad
> 

Hi,

Also soweit ich das sehe brauchen die Videostationen 26V als Spannungsversorgung. Das sollte aber kein Problem sein da die eh auf 3,3V gewandelt werden. Die Busleitung ist wohl weiterhin 24V.

Somit sollte das passen.
@peteh was meinst du dazu?

dannyyy, 2024-05-15:
@AzonInc Zielland wäre die Schweiz

dannyduando, 2024-05-15:
Hey @AzonInc ich hatte auf die Email geantwortet, aber ich weiß gar nicht ob das ankommt, ich hab meinen GitHub Account aufgemacht, weil ich das Projekt cool finde. 
Ich bräuchte die 9 Pin Variante und nach Deutschland, Danke dir!

AzonInc, 2024-05-15:
> Hey @AzonInc ich hatte auf die Email geantwortet, aber ich weiß gar nicht ob das ankommt, ich hab meinen GitHub Account aufgemacht, weil ich das Projekt cool finde. Ich bräuchte die 9 Pin Variante und nach Deutschland, Danke dir!

Da kam leider nichts an aber jetzt hab ich es hier mal nachgetragen :)

AzonInc, 2024-05-18:
@alkue1 du wolltest nur 3P Schraubklemme als minimalvariante oder?

alkue1, 2024-05-18:
Das sieht ja richtig gut aus! Ich bräuchte erstmal nur die 3 Pins, um das Format möglichst klein zu halten. Und im Zweifelsfall bekomme ich die Pins auch angelötet... Machst du den ESP mit einer Heizplatte oder Kolben fest? 
Bin sehr gespannt auf deine Tests:)

AzonInc, 2024-05-18:
> Machst du den ESP mit einer Heizplatte oder Kolben fest? Bin sehr gespannt auf deine Tests:)

Also in dem Fall werd ich es erstmal mit der Heizplatte versuchen, ansonsten Heißluft

AzonInc, 2024-05-18:
Testen konnte ich bisher LEDs, Relais, Spannungsversorgung, Brückengleichrichter.

Soweit ist alles bestens👌🏼

Die Buskommunikation kann ich erst testen wenn ich wieder in der Wohnung bin.

samdolt, 2024-05-19:
Ich bin auch interessiert. 9P und Schweiz. Ich habe dieses Model: https://www.kochag.ch/de/item/innensprechstelle-tc40-classic_DF41.7000

alkue1, 2024-05-19:
> Testen konnte ich bisher ESP, USB, LEDs, Relais, Spannungsversorgung, Brückengleichrichter. Soweit ist alles bestens👌🏼 Die Buskommunikation kann ich erst testen wenn ich wieder in der Wohnung bin.
> 
> Note
> 
> Nach abschließenden Tests in der Wohnung kann ich nun erfreut bestätigen dass alles genau so wie geplant ohne Einschränkungen und etwaige Störgeräusche funktioniert. Mit diesem Wissen kann ich jetzt weitere Exemplate bestellen.
> 
> Der Stückpreis beträgt aktuell **26€**
> 
> Zwecks Versand und Zahlung könnt ihr mir gern per Mail oder auf Discord per DM schreiben.
> 
> Bei denen von euch die 2 wollen wäre es ganz gut wenn man das am Ende bündelt um die Versandkosten bei einem Minimum zu behalten.

Dann bin ich sehr zuversichtlich, dass auch der Rest funktioniert. Starke Arbeit!

samdolt, 2024-05-21:
> Ich bin auch interessiert. 9P und Schweiz. Ich habe dieses Model: https://www.kochag.ch/de/item/innensprechstelle-tc40-classic_DF41.7000

Ich würde einen zweiten bestellen, ein Freund will auch einen. :-)

AzonInc, 2024-05-21:
@samdolt habs mit in die Liste genommen.

Und ins Gehäuse vom TCS ISH3030 passt es übrigens perfekt in die Ecke rein :)

AzonInc, 2024-05-21:
Die Bestellung der nächsten 5 habe ich eben auch aufgegeben :)
Das sollte also nicht mehr allzulange dauern.

NonProgrammerCharacter, 2024-05-22:
Moin, ich würde auch einen nehmen, 9P und nach DE.

Wlad2288, 2024-05-22:
> Hi,
>
> Also soweit ich das sehe brauchen die Videostationen 26V als Spannungsversorgung. Das sollte aber kein Problem sein da die eh auf 3,3V gewandelt werden. Die Busleitung ist wohl weiterhin 24V. [...]

Ok danke! ... dann würde ich auch einen nehmen.
Versand nach DE.

filo87, 2024-05-24:
Awesome project! Would love to get one board as well whenever the next batch is ready!

AzonInc, 2024-05-27:
@gee33 ich benötige von dir auch noch eine Adresse zwecks Versand.

dannyyy, 2024-05-27:
@AzonInc Bist du bereits am Versand der 1.4 Revision? *freu*

AzonInc, 2024-05-27:
> @AzonInc Bist du bereits am Versand der 1.4 Revision? 🤩

Die sollte Ende der Woche ggfs bei mir ankommen :)

AzonInc, 2024-05-28:
@dannyyy ich hatte dir bei Discord geschrieben - ich bräuchte auch noch eine Adresse :)

dannyyy, 2024-05-28:
> @dannyyy ich hatte dir bei Discord geschrieben - ich bräuchte auch noch eine Adresse :)

Habe dir eine Email gesendet (Adresse gemäss deinem Github Profil). Bin so gut wie nie auf Discord unterwegs.

azzieg, 2024-05-28:
This project is super-exciting. I would also like to order a board. I sent an email with details.

AzonInc, 2024-06-03:
@samdolt @NonProgrammerCharacter @Wlad2288 
Ihr könnt mir eure Adresse gern noch per Discord oder Mail an flo@azon.ai senden.
Da kann man dann auch alles weitere klären.

AzonInc, 2024-06-03:
Für die von euch die die a-Busleitung als Spannungsversorgung nutzen wollen habe ich leider eine schlechte Nachricht:
Ich habe es jetzt endlich testen können und es gibt leider ein piependes Störgeräusch auf die Leitung :(

Aktuell läuft es weiterhin nur mit P-Leitung oder externer Stromversorgung problemlos.

dannyyy, 2024-06-05:
@AzonInc habe heute dein Doorman erhalten und gleich installiert. Was soll ich sagen... absolut tolle Arbeit!! Alles funktionierte auf Anhieb. 
Wie würde sich die fehlende Leistung der Speisung bemerkbar machen? Ich habe das Gefühl, dass die Speisung die 60mA aufbringen kann. Aber das muss ich mal beobachten, ob es zu Reboots und Timeouts kommt.

Ich hätte gleich noch einen Vorschlag für ein nächstes Softwarerelease, bin mir nicht sicher, ob ESPHome das kann:
- Anstelle des Loggens und neu Flashens mit den korrekten Commands, könnte man diese nicht via Sensor Komponente oder Text Komponente in Home-Assistant gleich eintragen (manuell) und nutzen?
- Ich selbst habe keine ESPHome Integration in HA, weil mir dort die Möglichkeit für Add-Ons fehlt (Kubernetes). Ich habe das Update entsprechend mit `> esphome dashboard ./` gemacht, welches lokal das ESPHome Dashboard bereitstellt unter `http://localhost:6052`. Vielleicht könnte man das in der Anleitung noch ergänzen. Kann dir aber bis zum Wochenende auch ein PR machen mit einer ergänzenden Beschreibung und Screenshots

peteh, 2024-06-05:
> @AzonInc habe heute dein Doorman erhalten und gleich installiert. Was soll ich sagen... absolut tolle Arbeit!! Alles funktionierte auf Anhieb. Wie würde sich die fehlende Leistung der Speisung [...]

In meiner mqtt Firmware habe ich das tatsaechlich so umgesetzt, ist allerdings kein ESPhome. Aber eigentlich sollte es damit auch gehen. Im Prinzip ists ja nur eine int Zahl. 

Wenn man fuer den Text Sensor den Configuration Type setzt, dann taucht der auch nur in der Device Page auf. Ich speichere das dann in einem config file auf dem esp32.

AzonInc, 2024-06-06:
@dannyyy bei Gelegenheit werde ich nochmal schauen ob ich es vereinfachen kann :)

Und zwecks Anleitung kannst du gern etwas auf Basis vom dev Branch hinzufügen

K-r-i-s-t-i-a-n, 2024-06-08:
Sehr interessantes Projekt, bitte mich auch auf die Liste setzten. Details habe ich per Email geschickt.

AzonInc, 2024-06-13:
@zopieux did you receive my email regarding the wires? Maybe there is still a wire with 24V hidden behind.
By the way: I can't receive mails at the moment because the mailserver is down.

In case you're not interested anymore just let me know so I can continue with shipping to switzerland :)

zopieux, 2024-06-14:
Apologies, I won't be at my place until Monday to reply. I also access to the PSG02 in the common areas, I'll check if the P wire is even connected there. I do in fact have 4 wires coming out the wall but one is for the door bell, and I suspect the "P" wire is just the door bell return wire. I'll check the potential between b and P. The PCB is configured in two-wire mode, that much I checked, but maybe there is more hidden wire in the wall.

AzonInc, 2024-06-14:
No problem.

> one is for the door bell of course.

Yea actually two of them are connected to the apartment doorbell button.

zopieux, 2024-06-14:
Yeah, I suppose the P screw is the bell return wire, not the 24V :(

zopieux, 2024-06-19:
No dice. :frowning_face:  While the P wire is indeed connected on the BVS20, whatever arrives at my place is just a two-wire cable, and I confirm the other two are the door bell button.

>there is a beeping noise interference on the line
>only operates smoothly with P

I'm not sure I understand what that means. Is the point that we can _try_ running the ESP from just a+b, but it introduces noise in the a+b comms? Are you talking about the voice transport (affecting all residents), or the protocol messages being unreliable?

If it means e.g. the software needs to send messages a few time before it goes through, I think I can live with that. However if you're saying this makes my neighbors' intercom noisy, that's obviously a no go. :-)

Thanks for testing!

matejbart, 2024-06-29:
@AzonInc Bitte 1x die 9P Variante (DE), danke.

AzonInc, 2024-07-11:
@smartforcedev lebst du noch? :)

BiasF, 2024-08-05:
Moin,

Wir hatten grade per E-Mail kontakt, kannst mich also auf die Bestellliste setzen 

Ich hab grade bei mir die Station mal aufgemacht und gesehen das alle 4 Anschlüsse belegt sind. Also A, B, E & P. 
Macht das einen Unterschied?

[image]

AzonInc, 2024-08-05:
> Moin,
> 
> Wir hatten grade per E-Mail kontakt, kannst mich also auf die Bestellliste setzen 
> 
> Ich hab grade bei mir die Station mal aufgemacht und gesehen das alle 4 Anschlüsse belegt sind. Also A, B, E & P. 
> Macht das einen Unterschied?
> 
> [image]

Das sollte passen. An E&P hängt wahrscheinlich der Etagen Klingeltaster von der Wohnung wie bei mir.

BiasF, 2024-08-05:
Ich messe da 1,2V bei 0,81mA.
Kommt mir ein bisschen wenig vor.

protyposis, 2024-08-08:
Received my Doorman today, installed it into TSC ISH3030, and works nicely👍 Great hardware build quality, stable firmware, simple config - thanks @AzonInc!

diaznet, 2024-12-06:
Installed with 3 wires on TC40 and works like a charm, thanks.

closed 2025-02-11 by AzonInc