AzonInc/Doorman #5: Feature: Audio Support

AzonInc, 2024-05-29:
It would be a nice addition to add audio support to the board as well.
If you have any knowledge about audio processing or ideas how to send audio on the bus feel free to share it here.

I've been able to enable and disable the audio mode already but now we need to process the data somehow as a first step.

BiasF, 2024-08-08:
I tried ring intercom last year, it worked but the audio quality was very bad with a high frequent beep in the background.
Didn't feel comfortable with the cloud connection and sent the device back, but it shows that its possible to build that feature.

artifactdev, 2024-11-29:
Might it be possible to "debug" what the ring intercom does, when enabling audio and reverse engineer it? Especially as it is available pretty cheap right now. But as I have it here already and will get a doorman soon, I might be able to support on this.

artifactdev, 2024-12-03:
Here someone also played with audio on tcs bus a bit already, maybe worth a watch even if its some older: https://www.youtube.com/watch?v=TMwRrWqhUkE

AzonInc, 2024-12-03:
Mhh I'm just not sure how to debug it.
I will have an oscilloscope in the next days, sooo maybe I'll find out something new (after figuring out how to use that oscilloscope haha).

Let me know when u find out anything new :)

colinmarc, 2025-02-14:
It seems like it's possible to compile libopus for the esp32 (see for example https://github.com/XasWorks/esp-libopus). One could feed it from an ADC, maybe one of these:

https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC

AzonInc, 2025-02-14:
Would it be possible to work with the raw 24V data?
The audio stream is somewhat modulated on the DC but I'm not sure, I couldn't measure any AC.
It's ~22V DC base and ~1-2V audio on top

colinmarc, 2025-02-14:
Disclaimer: I have no idea what I'm talking about

I think what's need is an op-amp to measure the difference between the audio signal and the 22v baseline, and then amplify it to the range that the ADC wants (usually 5v or so?). See "difference amplifier" here https://forum.arduino.cc/t/using-operational-amplifiers-in-your-arduino-project/692648/4

AzonInc, 2025-06-21:
I just found this:
https://electronics.stackexchange.com/a/692106/580470

Somebody created the schematic to achieve the audio reading. However I am also not a professional so I don't know how to implement that properly.

2nv2u, 2025-07-12:
Although I have no clue how and if even feasible, but it would be great to add video support (2 channels for 2 door systems) as well using ONVIF Profile S (RTSP, also for audio), perhaps even support the Profile A for controlling the door locks.
https://www.onvif.org
https://www.home-assistant.io/integrations/onvif
https://en.wikipedia.org/wiki/Real-Time_Streaming_Protocol

AzonInc, 2026-02-05:
Update:
We evaluated audio circuits that use coupling capacitors to transmit and receive audio over the bus.
A new PCB revision is currently under development, designed to be extensible via a dedicated audio expansion board.

kroimon, 2026-02-21:
I have an intercom by SKS Kinkel and reverse engineered their bus to be very similar. The audio is superimposed onto the 19-28 VDC baseline. The control messages are using FSK (Frequency Shift Keying), so basically sending two different tones for 0 and 1 instead of just dropping the voltage.
I would love to get my hands on a Doorman with audio functionality that I could use to implement the protocol for ESPHome.

Alexandero89, 2026-09-25:
Hey @AzonInc ,
sorry to stress you on this one, but is there any progress on this?
Would be really awesome to have this feature implemented.
Thanks in regard!