AzonInc/Doorman #10: Buying doorman

gregmag, 2024-07-04:
### Description

Hi, i am interested to buy a Doorman S3 ...
Can you send it to switzerland ? How much does it costs ?
Thank you and congratulation for this amazing developement ?

### How do you manage your Doorman's updates?

None

AzonInc, 2024-07-07:
Hi, thanks for ur interest. Just send me a mail to discuss further details :)
flo@azon.ai

closed 2024-07-07 by AzonInc

AzonInc, 2024-07-18:
Are you still interested?

pumabaer, 2024-07-25:
Hello, I would also be interested in a circuit board. 
What do you want for it?

AzonInc, 2024-07-27:
Hi,

for a the fully assembled Doorman it's 30€ + shipping.

Best regards
Flo