AzonInc/Doorman #13: Ich messe da 1,2V bei 0,81mA.

AzonInc, 2024-08-05:
Ich messe da 1,2V bei 0,81mA.
Kommt mir ein bisschen wenig vor. 

_Originally posted by @BiasF in https://github.com/AzonInc/Doorman/issues/1#issuecomment-2269732936_

Das ist durchaus etwas wenig.
Und wie schaut's auf den anderen Leitungen aus?

BiasF, 2024-08-05:
Ich messe gleich mal gegen A, normal müsste die Spannung ja bei 24v liegen.

BiasF, 2024-08-06:
Zwischen A und P beträgt die Spannung 22,8V, kommt schon eher hin

AzonInc, 2024-08-06:
Okay und zwischen a und b müssen es auch ähnlich viel sein wie a und p

BiasF, 2024-08-06:
jop, da sind es 23,8.

AzonInc, 2024-08-06:
Dann ist alles kompatibel und die Leitungen nur vertauscht. Die Doorman Platine dreht sich die eh so wie sie gebraucht werden.
Da klemmst du dann a & b je an ein TCS:Bus und P an P.

BiasF, 2024-08-06:
TOP dann bin ich schon sehr gespannt ob alles funktioniert :)

AzonInc, 2024-09-07:
Ein Workaround für das Piepen wäre ein 56 oder 100 Ohm Widerstand als Brücke zwischen P und Bus +24V

AzonInc, 2024-09-07:
[image]
Das ganze würde dann so aussehen.

BiasF, 2024-09-07:
Das teste ich mal in den nächsten Tagen wenn ich wieder daheim bin, so einen Widerstand müsste ich noch rumfliegen haben 👍

AzonInc, 2024-09-07:
mit 100 peipt es nicht mehr aber es kommen auch keine commands rein weil die spannung wahrscheinlich davon zu beeinträchtigt ist
mit 56 piept es auch nicht aber damit könnten wieder commands rein (muss ich selbst mal testen)

AzonInc, 2024-09-07:
Okay also das läuft nicht wie gewünscht.
Stromversorgung ohne große Störgeräusche ja, Daten lesen nein 🥲

AzonInc, 2024-12-06:
Moin, wie betreibst du das Teil derzeit? Wahrscheinlich mit externer Spannungsversorgung oder?

BiasF, 2025-02-09:
Moin,

Ich habs aktuell einfach per USB Angeschlossen.

AzonInc, 2025-02-11:
Okay, denke dann passt das erstmal. Ich habe leider erst mit rev 1.5 eine mehr oder weniger schöne lösung gefunden

closed 2025-02-11 by AzonInc