AzonInc/Doorman #24: Feature: Second Doorbell

m4n0v31, 2024-09-28:
### Description

Hi,
I received and installed the doorman3 and it is grerat.
I've the following question.

my building has a setup a little different:
- Entrance Door: with Doorbell and Camera
- Second Door: with Doorbell and Camera
- Apartment Door: with Doorbell

The system has:
- command to open entrance door (already supported in the stock configuration)
- command to open second door (already supported in the stock configuration)
- command for doorbell at entrance (already supported in the stock configuration)
- command for doorbell at second door (not available in stock configuration)
- command to pick up phone at entrance, show video at entrance (already supported in the stock configuration)
- command to pick up phone at second door, show video at entrance
- command to hang up at entrance
- command to hang up at second door
- command to switch from entrance video to second door video

There is only one button to pick up phone, and it starts showing the video last seen or where the last doorbell was rang. So if I press the button it could be any of the two command picking up the phone.

I would like to : 
- be able to configure the command for the doorbell at second door and show this as an entity (in the same way "apartment doorbell" and "entrance doorbell" are made)
- be able to detect and react on pattern at second doorbell
- be able to have "ring to open" active also at second doorbell, or have the possibility of configure "ring to open" so that the entrance door is open and after 10s the second door is open.
- be able to insert two command for the phone pick up (any of the two is valid as phone pick up)
- be able to insert two command for the phone hand up (any of the two is valid as phone hang up)

What's the best way to support this setup?

### Which version of the ESPHome Doorman firmware config are you running?

Firmware: 2024.9.1

### How do you manage your Doorman's updates?

None

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2024.9.1

AzonInc, 2024-09-29:
Are you running the latest Dev firmware or stable one?

m4n0v31, 2024-09-30:
I'm running stable. 
I'll try out dev and let you know. Thx

AzonInc, 2024-09-30:
> I'm running stable. 
> I'll try out dev and let you know. Thx

But please note that the Dev version is mostly a rewrite and works differently.

It doesn't really use commands in the way stable used them. So you have to set it up again.

Please follow the getting started guide of the Dev version. It has all the information you will need.

m4n0v31, 2024-09-30:
OK,
I tried to switch to dev by changing the package in the doorman-s3.yaml file from esphome to:
`AzonInc.Doorman: github://AzonInc/doorman/firmware/doorman-stock.yaml@dev`

but I get the following:
```
Failed config

number.template: [source /config/.esphome/packages/efae6b5d/firmware/addons/memory-utils.yaml:22]
  
  optimistic cannot be used with lambda.
```

AzonInc, 2024-09-30:
> but I get the following:

Should be fixed now

m4n0v31, 2024-09-30:
On the other side. How long until the changes I'm interested in make it to "stable"?
Is it worth to switch to dev and then stay there?

AzonInc, 2024-09-30:
> On the other side. How long until the changes I'm interested in make it to "stable"?
> Is it worth to switch to dev and then stay there?

Maybe two or three more weeks when im Back in Germany with more time.

The dev version improves the Plug & Play experience a lot as no commands are needed and the TCS protocol is implemented to some extent. But also has some breaking changes due to removal or renaming some entities.

I will write everything down(Discord announcements) in detail when it's ready.

m4n0v31, 2024-09-30:
do you know if it is possible to override the "initial_value" of the globals in the yaml file, so I can already put in my commands and don't care if I switch from one branch to the other?

AzonInc, 2024-09-30:
Thats not gonna work as these commands are subject of removal.

They dont exist in dev.

AzonInc, 2024-09-30:
The new system is using serial numbers only and is saving them separately.

AzonInc, 2024-09-30:
I just looked at the code and it's like that:
There is a pick up phone sensor which triggers whenever you talk to the entrance or second door.
So both of them trigger the pickup phone pattern.

It's the same behaviour for the hang up phone sensor. it doesn't matter which station youre talking to. Whenever you hang up the phone it will trigger.

The second doorbell sensor triggers the same doorbell pattern event entity.
But it uses the second_ event types:
      - "apartment_single"
      - "apartment_double"
      - "apartment_triple"
      - "entrance_single"
      - "entrance_double"
      - "entrance_triple"
      - "second_entrance_single"
      - "second_entrance_double"
      - "second_entrance_triple"
      
      
      
So based on your request the following is already in dev:
- be able to configure the command for the doorbell at second door and show this as an entity (in the same way "apartment doorbell" and "entrance doorbell" are made)
- be able to detect and react on pattern at second doorbell
- be able to insert two command for the phone pick up (any of the two is valid as phone pick up)
- be able to insert two command for the phone hand up (any of the two is valid as phone hang up)

This needs to be done:
- be able to have "ring to open" active also at second doorbell, or have the possibility of configure "ring to open" so that the entrance door is open and after 10s the second door is open.

AzonInc, 2024-11-16:
Please check the latest release and open a new issue if there is still something missing.

closed 2024-11-16 by AzonInc