AzonInc/Doorman #31: open_door command always uses persisted serial_number

matejbart, 2024-11-23:
### Description

Hi @AzonInc ,

I have updated to the latest v1.5.0 and the `open_door` command no longer works (for my outer station) because the command is now built differently (with the internal station's serial_number):

https://github.com/AzonInc/Doorman/blob/2340d89c2fc572b41363b56cfcb34ecd43f633d2/components/tc_bus/protocol.cpp#L63-L77

The firmware [always tries](https://github.com/AzonInc/Doorman/blob/2340d89c2fc572b41363b56cfcb34ecd43f633d2/components/tc_bus/tc_bus.cpp#L592-L596) to pass the persisted `serial_number` to the buildCommand (if it's not set by the caller or if it's just 0).

I therefore modified the `protocol.cpp` by building the `open_door` without the `serial_number` and that solves the problem for me.

I'm just curious why the composition of that command has changed, is this necessary for any other model types? (I'm using a `ISH 1030` internal station).

Another thing that I have noticed is the last byte of the open_door command: In the case without the `serial_number` the address has no longer the leading `8`.

For example, my internal station always sends `1180`, but that is no longer possible (even though `1100` works just fine). However, since `COMMAND_TYPE_STOP_TALKING_DOOR_CALL` also uses the `8`, I was wondering if this should be aligned with the open_door cmd as well.

Best regards,
Matej

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2024.11.1

matejbart, 2024-11-24:
> I just noticed that this could be a streamlined way of building this command while looking at the protocol definitions.

Are you referring to the cmd `langer Türöffner` from the [protocol definitions](https://github.com/user-attachments/files/17893181/TCSBusProt.xml.zip) (starting from line 250)?

From my humble understanding I believe this command serves a different purpose, maybe to keep doors opened until another command is issued (even though I couldn't find such a command). However, in my case the outdoor station uses a door opener relay ("buzzer") to open the door, so I think that the `langer Türöffner` is simply not designed to work with this setup. (But please correct me if I'm wrong, maybe this works at your outdoor station?)

Furthermore, as all the other indoor stations always just use the `11XX` command, I would basically prefer to stick with that and try to mimic the behavior of the indoor stations as good as possible.

> I couldn't find any definition of that byte in the protocol definitions. That's why I didn't implement it.

It seems that an `8` (in hex) denotes that the address is coming or targeting an outdoor unit, and when it's `4` then it's from an indoor station. It's the bit before the 6-bit address (definitions starting from line 139 for example, but this is also valid for the `3xxx` cmds as well).

> We could just remove the long version of the open door command or include a flag to use the king version but when it's not set always use the short 11xx version.

Sounds good, I just think that using the "long" version should probably not be the default option.

> Another question: Could it be related to your outdoor station changing its address? #32 

No, the "changing address" thing happened afterwards.

> But as u send 1100 with address 0, which works, it's not related to the address.

It's really interesting that even though the outer stations address has changed, sending `1100`, `1180` and `1181` still works. Maybe there is a setting in the outdoor stations unit to always react on 11xx`?

AzonInc, 2024-11-24:
> Are you referring to the cmd langer Türöffner from the protocol definitions (starting from line 250)?

Yea, that's the one. Here in this building they also have the door buzzer and probably the same control unit as in urs (there is only option as far as I know). Here it worked just fine (also at other peoples setup) so I thought implementing it is fine but I will remove the serial number version.

> It seems that an 8 (in hex) denotes that the address is coming or targeting an outdoor unit, and when it's 4 then it's from an indoor station. It's the bit before the 6-bit address (definitions starting from line 139 for example, but this is also valid for the 3xxx cmds as well).

I'm not sure if an outdoor station is opening the door itself (at least I've never seen an outdoor station sending 11xx by itself, but never say never).
For the 3xxx its just the door readiness flag to pickup calls (after ringing)

> It's really interesting that even though the outer stations address has changed, sending 1100, 1180 and 1181 still works. Maybe there is a setting in the outdoor stations unit to always react on 11xx`?

If there are two outdoor stations 1100 won't work for both of them (tested).

AzonInc, 2024-11-24:
It's changed in [2024.11.2](https://github.com/AzonInc/Doorman/releases/tag/2024.11.2)

AzonInc, 2024-11-25:
Can I close this or do we still need to do smth here?

matejbart, 2024-11-25:
Thx for the update, and yes I have also seen in the manual of the service device that there is a difference between the short and long open door command, but I couldn't figure out which command to send to check if the limitation on the outer station is active or not, maybe this is something that can be only programmed via the configo software?

AzonInc, 2024-11-25:
> Thx for the update, and yes I have also seen in the manual of the service device that there is a difference between the short and long open door command, but I couldn't figure out which command to [...]

Yes that's what the guy told me today, it's an advanced feature only configurable via configo.

matejbart, 2024-11-26:
Just for the sake of documentation, the [online configo manual](http://tcs-forum.de/download/download_tcsag/configo_Online-Hilfe/#8) reveals that the setting `Lange Türöffnerfunktion als Rufprotokoll` is set on the control unit (and not the outer unit as I have wrongly guessed before), it would be just interesting to figure out how the internal stations know that the long command should be used - this could be also tracked by the firmware and adjusted automatically.

Another thing worth noting is that the [outer stations](http://tcs-forum.de/download/download_tcsag/configo_Online-Hilfe/#6.1) have a setting to only allow the open door command to be "executed" while being in `Türbereitschaft` or even only during a voice call, but this seems to be deactivated by default.

AzonInc, 2024-11-26:
> Another thing worth noting is that the outer stations have a setting to only allow the open door command to be "executed" while being in Türbereitschaft or even only during a voice call, but this seems to be deactivated by default.

Some have it enabled which is quite annoying because it prevents the open door button to work properly.

> it would be just interesting to figure out how the internal stations know that the long command should be used - this could be also tracked by the firmware and adjusted automatically.

I have a TCS test system at home so I could try whatever but I just need to know what.

AzonInc, 2024-11-27:
It somehow "just works" without further doing

As you have mentioned in #32 (comment) that your TCS system starts to use the long_door command after the control button of any internal station is hit. Does every other indoor station react on that command immediately, or does it need to be called out by the outer station first?
Because if it's the latter, then this information - that the indoor stations need to use the long_door command - must be somehow encoded in the door call command.

[
{
"target": "https://github.com/AzonInc/Doorman/issues/31#issuecomment-2504203503",
"url": "https://github.com/AzonInc/Doorman/issues/31#issuecomment-2504203503",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

matejbart, 2024-11-27:
ok, then I think we're done here 👍 feel free to close the issue.

closed 2025-01-12 by AzonInc