AzonInc/Doorman #38: Either main OR second entrance is not recognized after setup mode

falschparker82, 2025-01-11:
### Description

I've tried setup mode multiple times (after each one resetting the Doorman v1.5 to factory defaults). The problem is that just one of the doors is ever recognized — interestingly enough, both as "Second Entrance" — and the other one ignored. I just see the other door as bus events. Here are the corresponding logs. I guess something doesn't quite work with guessing the second entrance ID?

Below are the corresponding bus events. In the meantime, is there any way to hardcode both addresses, and if yes, which part is the ID? (I guess the middle hex part?)

Help very much appreciated.

### Logs

```plain text
Main Entrance:
changed to 0E88C683
14:30:41 - 9 minutes ago
changed to 1E88C683
14:23:08 - 17 minutes ago

Second Entrance:
changed to 0E88C681
14:34:21 - 5 minutes ago
changed to 1E88C681
14:23:34 - 16 minutes ago

Some misc bus events of questionable relevance (I don't speak TCS bus just yet :)
changed to 00008414
14:36:24 - 15 minutes ago
changed to 00008410
14:36:24 - 15 minutes ago
changed to 0000840C
14:36:24 - 15 minutes ago
changed to 00008408
14:36:24 - 15 minutes ago
changed to 00008404
14:36:23 - 15 minutes ago
changed to 00008400
14:36:23 - 15 minutes ago
changed to 810E88C6
14:36:23 - 15 minutes ago
changed to 00005800
14:36:23 - 15 minutes ago
changed to 00002481
14:35:29 - 16 minutes ago
changed to 00002281
14:34:33 - 17 minutes ago
changed to 0E88C681
14:34:29 - 17 minutes ago
changed to 00002281
14:34:21 - 17 minutes ago
changed to 0E88C681
14:34:21 - 17 minutes ago
```

### Which version of the ESPHome Doorman firmware config are you running?

? (Hardware Doorman-S3 1.5.0) -> good idea to show FW version on web interface?

### How do you manage your Doorman's updates?

OTA Updates

AzonInc, 2025-01-11:
Hi, could you please send me the webserver logs?
But you need to set loglevel do DEBUG first. Please compile the firmware using
```
substitutions:
  log_level: "DEBUG"
```

The debug logs contain everything that I need to know :) Also the decoded commands including AS address and payloads.

falschparker82, 2025-01-12:
Hi Flo, here are the logs:

```
## MAIN ENTRANCE
[17:05:36][D][sensor:093]: 'Bus Voltage': Sending state 3.02800 V with 2 decimals of accuracy
[17:05:37][D][tc_bus:246]: Received command 0E88C683
[17:05:37][D][tc_bus:441]: [Parsed] Type: DOOR_CALL, Address: 3, Payload: 0, Serial: 952518
[17:05:37][D][text_sensor:064]: 'Last Bus Command': Sending state '0E88C683'
[17:05:37][D][binary_sensor:036]: 'Second Entrance Doorbell': Sending state ON
[17:05:37][D][tc_bus:572]: Send event to Home Assistant on esphome.doorman
[17:05:37][D][binary_sensor:036]: 'Second Entrance Doorbell': Sending state OFF
[17:05:37][D][sensor:093]: 'Bus Voltage': Sending state 2.98400 V with 2 decimals of accuracy
[17:05:38][D][sensor:093]: 'Bus Voltage': Sending state 3.01200 V with 2 decimals of accuracy
[17:05:39][D][event:015]: 'Doorbell Pattern' Triggered event 'second_entrance_single'
[17:05:39][D][tc_bus:611]: Sending command 00001103
[17:05:39][D][tc_bus:441]: [Parsed] Type: OPEN_DOOR, Address: 3, Payload: 0, Serial: 0
[17:05:39][D][text_sensor:064]: 'Last Bus Command': Sending state '00001103'

Second Outdoor Station after Setup:
Serial: 952518
Hex: E88C6
ID: 3

## SECOND ENTRANCE
[17:05:59][D][sensor:093]: 'Bus Voltage': Sending state 3.02000 V with 2 decimals of accuracy
[17:06:00][D][sensor:093]: 'Bus Voltage': Sending state 3.00000 V with 2 decimals of accuracy
[17:06:00][D][tc_bus:246]: Received command 0E88C681
[17:06:00][D][tc_bus:441]: [Parsed] Type: DOOR_CALL, Address: 1, Payload: 0, Serial: 952518
[17:06:00][D][text_sensor:064]: 'Last Bus Command': Sending state '0E88C681'
[17:06:00][D][tc_bus:572]: Send event to Home Assistant on esphome.doorman
[17:06:01][D][sensor:093]: 'Bus Voltage': Sending state 3.00000 V with 2 decimals of accuracy
[17:06:02][D][sensor:093]: 'Bus Voltage': Sending state 3.00300 V with 2 decimals of accuracy
[17:06:03][D][sensor:093]: 'Bus Voltage': Sending state 3.01500 V with 2 decimals of accuracy
[17:06:04][D][sensor:093]: 'Bus Voltage': Sending state 3.01700 V with 2 decimals of accuracy
[17:06:05][D][tc_bus:246]: Received command 00002281
[17:06:05][D][tc_bus:441]: [Parsed] Type: END_OF_RINGTONE, Address: 1, Payload: 0, Serial: 0
[17:06:05][D][text_sensor:064]: 'Last Bus Command': Sending state '00002281'
[17:06:05][D][tc_bus:572]: Send event to Home Assistant on esphome.doorman

Second Outdoor Station after Setup:
Serial: 952518
Hex: E88C6
ID: 1
```

falschparker82, 2025-01-12:
From the base.yaml I assume all of this is convention over configuration at ID = 0 and my main entrance ID = 3 seems pretty nonstandard, right?

```
# Preconfigured Open Door Button
  - platform: template
    id: open_entrance_door
    name: "Open Entrance Door"
    icon: mdi:door
    on_press:
      - tc_bus.send:
          type: open_door
          **address: 0**
    web_server:
      sorting_group_id: sorting_group_controls
      sorting_weight: -10
 ```

AzonInc, 2025-01-12:
Fixed in dev branch