AzonInc/Doorman #39: Disable Bluetooth

protyposis, 2025-01-12:
### Description

Hi, I'm running firmware 2024.11.2 and noticed that "Doorman-s3" shows up as an available Bluetooth device on my phone, even though my setup is completed. Is this expected behavior, and if so, what's the use case? Would it make sense to make it configurable with a system setting?

I'm not sure if I remember correctly, but I believe older firmware versions turned it off after the setup was completed, so I'm not sure whether this is a bug report or a feature request.

AzonInc, 2025-01-12:
Mhh I'm not aware of this issue.
Maybe the latest ESPHome version has a bug or smth.

protyposis, 2025-01-13:
Ok, so you're saying this is unexpected behavior.

I originally used ESPHome 2024.11.3 and just reinstalled with 2024.12.2 - Bluetooth still active.

AzonInc, 2025-01-13:
Maybe nobody ever noticed this behaviour.

I just tried to look for Bluetooth devices on my phone but can't seem to find any around here.

Is there anything in the logs about improv ble?
And does Home Assistant still discover the device?

matejbart, 2025-01-13:
Hi,

I have also noticed that (and thought it's default behavior), therefore I disabled ble improv by commenting `esp32_improv:` in the `doorman-stock.yaml`. 

Regardless of whether this is a bug or should be addressed in some other way, I want to point out that the [esp32_improv doc](https://esphome.io/components/esp32_improv) also includes the following warning:

> The BLE software stack on the ESP32 consumes a significant amount of RAM on the device.

So maybe it might be worth considering to disable ble improv by default?

AzonInc, 2025-01-13:
It's in there as an easy way to setup the device for the first time.
But in the Nuki Bridge Firmware its disabled as it's incompatible with the NimBLE Stack used by the Nuki component.

Not sure how often it's used but mDNS doesnt work in some cases so I thought BLE Improv would be a nice addition.

closed 2025-01-13 by protyposis

AzonInc, 2025-01-13:
FYI: It's fixed in the dev channel. The issue is not improv, but the BLE server itself which is automatically started when using the improv component. The BLE Server will now stop as soon as wifi is connected and will start when wifi is disconnected.