AzonInc/Doorman #41: Update auf 25.02 nicht möglich

BiasF, 2025-02-09:
### Description

Moin,

nach dem ich mich länger nicht mit dem Gerät beschäftigt hab (lief halt), hab ich dieses Wochenende etwas mehr rumgespielt.
Ich hab also über ESP-Home Dashboard  aktualisiert, danach hing er im setup mode und ich kam nich mehr dran über ESP-Home.
Ich hab dann gesehen das etwas größere Änderungen durchgeführt wurden, hab also über USB-C  mal komplett neu geflashed, inklusive erase. 

Vorher aus dem ESP-Home Dashboard entfernt und auch das Gerät aus HASS entfernt, damit sich da nix in die Quere kommt.

Nun hänge ich trotz dem wieder im Setup, komme auch nicht über das AP Wifi auf den ESP, scheitert am PSK den er nicht akzeptiert. 

Hab mal logs hochgeladen. Ich hoffe ich hab ihn nicht zerstört ^^

### Logs

```plain text
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/18726110/esp-web-tools-logs.txt)
```

### Which version of the ESPHome Doorman firmware config are you running?

2025.2.0

### How do you manage your Doorman's updates?

ESPHome YAML Config

### If you're using an ESPHome YAML config, which version of ESPHome are you running?

2025.2.0

### ESPHome Config

```YAML

```

AzonInc, 2025-02-09:
Flash mal bitte per dev docs Seite 
https://doorman-dev.surge.sh/guide/firmware/stock-firmware

BiasF, 2025-02-09:
2025.2.1 is noch schlimmer ^^

[esp-web-tools-logs_2025.2.1.txt](https://github.com/user-attachments/files/18726154/esp-web-tools-logs_2025.2.1.txt)

AzonInc, 2025-02-09:
Ahh du hast rev 1.4 da sind nur 2mb PSRAM teilweise drin.
Füg das mal bitte zu esphome dashboard hinzu und nutze folgende config:
```
substitutions:
  name: "doorman-s3"
  friendly_name: "Doorman S3"

# Import Doorman Stock Firmware Config
packages:
  AzonInc.Doorman: github://azoninc/doorman/firmware/ha-doorman-stock.dev.yaml@dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

psram:
  mode: quad
```

BiasF, 2025-02-09:
probier ich grade, allerdings is mein ESPHome Dashboard noch auf ESPHome 2024.12, ne neuere version is gar nicht (stable) released.

Da bemängelt er dann das meine ESPHome Bersion zu alt sei um das zu kompilieren.

AzonInc, 2025-02-09:
Stimmt, da brauchst das esphome dev addon für

AzonInc, 2025-02-11:
Ah sorry, hab die config oben nochmal aktualisiert. Teste bitte nochmal.

BiasF, 2025-02-12:
Leider lässt es sich immer noch nich flashen.
Hab mir nun auch das ESPHome Dev Addon installiert. 

[logs_doorman_run.txt](https://github.com/user-attachments/files/18770060/logs_doorman_run.txt)

AzonInc, 2025-02-12:
Okay sehr gut, für zukünftige updates solltest du nun auch jeweils die richtige version bekommen

BiasF, 2025-02-12:
Top, danke !

Was mir grade noch aufgefallen is, der Doorman erkennt bei mir das Klingeln gar nicht. 
Ich bin mir grade aber nicht sicher, ob es mit der alten Firmware ging. Habe das Feature irgendwie vernachlässigt und nie getestet ^^

AzonInc, 2025-02-12:
öh also du kannst das ganze mal mit einer alten version testen:

```
# Doorman S3 Stock Firmware

# You can change a few options here.
substitutions:
  name: "doorman-s3"
  friendly_name: "Doorman S3"

# Import Doorman Stock Firmware Config
packages:
  AzonInc.Doorman: github://AzonInc/doorman/firmware/doorman-stock.yaml@2024.11.2

external_components:
  - source: github://azoninc/doorman@2024.11.2
    components: [ tc_bus ]
    refresh: 60s

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Das benötigt esphome stable 2024.12.X