AzonInc/Doorman #45: Koch Ecoos Support

yannikgold, 2025-03-09:
Based on the documentation, it appears that Koch Ecoos is not currently supported. Is that correct, or could you clarify what is meant by ‘Verification and implementation required’? My system is a Koch Ecoos, and I could test its functionalities if I can obtain the necessary hardware. Additionally, I may be able to assist with the implementation. My primary goal is to enable remote door unlocking (preferably with mobile notifications), or if that proves too complex, to at least automate door unlocking when the doorbell rings.

AzonInc, 2025-03-10:
The list is just for settings availability.
In general all models are supported, but settings are only available for a subset of them. Also not every setting is available for every model :)

AzonInc, 2025-03-12:
If you want to order one, just send me an email or dm me on Discord :)

closed 2025-05-19 by AzonInc