AzonInc/Doorman #46: Homekit

findthebug, 2025-03-10:
### Description

Support to connect the door opener action to homekit would be super.

Since this is using ESP maybe there is a way to integrate HomeSpan, so the actions like doorbell, opener are exposed to homekit.

looking for a way to controll Koch TC50, door opener with homekit. This project looks promising to me. 

also using Homebridge, maybe there is a way to connect Doorman with a plugin there?

findthebug, 2025-03-14:
if i see this correctly. there is a firmware for doorman & MQTT only right? Homebridge has also some MQTT plugins, maybe it's even easier to communicate directly with doorman?

AzonInc, 2025-03-14:
Yea I saw that as well but you would need to add the entire topic structure to your config and build it.

findthebug, 2025-03-14:
at the end all im looking for is a way to trigger door opener in some way. since the doorman has an IP address, is there a chance to just do some simple curl to trigger door opener?

AzonInc, 2025-03-14:
Yes of course, you can use the ESPHome API and make a call to toggle the button.

findthebug, 2025-03-14:
Ah ok. Do i need Home Assistant setup for this? Or is there a firmware i can implement this without HA?

AzonInc, 2025-03-14:
You don't need Home Assistant for it, just use the custom config and add the ESPHome API again. If you need an example I can help you.

findthebug, 2025-03-14:
A ok all good. Will check all of this as soon as i have the device. Thanks!

findthebug, 2025-04-27:
Got the Device and all was working in less then 10 minutes!

1. Did the stock firmware update to 2025.3.0
2. Did the default device setup wlan & wiring on a Koch device
3. Checked the web interface to test the device via "Open Entrance Door", which was working first press!
4. Now i use "curl  http://doorman-s3/button/open_entrance_door/press" in homebridge to trigger the action

Thanks 100times!

closed 2025-04-27 by findthebug

AzonInc, 2025-06-11:
I worked on native HomeKit support for doorbell events and door locks.
It's available in the dev branch. Its a special variant that has to be flashed.

https://dev.doorman.azon.ai/guide/firmware/installation

[image]

https://dev.doorman.azon.ai/guide/firmware/homekit