AzonInc/Doorman #47: Station with two Doors

lukaiser, 2025-05-14:
### Description

Our second station has two connected doors. You can ring all apartments from the station but depending on the indoor station one or the other door opens. Therefore, the short code e.g. „1102“ doesn’t do the trick, it needs to be the long code e.g. “16CFEA82”.

Thanks for considering the feature!
Lukas

lukaiser, 2025-05-16:
Apparently, the long code isn't enough to open the correct door. It also seems to depend on beforehand used doorbell. Does anybody know if there is a code that might work?

lukaiser, 2025-05-18:
The system seems to work as follows:
When pressing a doorbell button, the second station sends two ring commands. One to the indoor station and another ring command to one of the two [relay](https://www.tcsag.de/fileadmin/user_upload/TCS_DE/Metanavigation/Downloads/Produktinformationen/Funktionserweiterungen/PI_BRE2-SG.pdf).
The second ring command seems to prime the relay for 30 seconds.
Sending the ring command to the relay and half a second later the open door command seems to work reliable to open the correct door.
I am still trying to figure out, if there is a command to open the relay directly but haven't found anything useful yet. Any idea? Thanks for the help!

AzonInc, 2025-05-18:
Okay, could you provide the exact commands / logs when you open each door?

lukaiser, 2025-05-18:
### Log using the original system:

**Apartment 1**

1. Pressing Bell
2. 0007D082 -> Bell to Relay Door 1
3. 06CFEA82 -> Bell to Indoor Station 1
4. Pressing Door Opener
5. 16CFEA82
6. 2282
7. 2482

**Apartment 2**

1. Pressing Bell
2. 0003E882 -> Bell to Relay Door 2
3. 06CFD182 -> Bell to Indoor Station 2
4. Pressing Door Opener
5. 16CFD182
6. 2282
7. 2482

### Using Doorman Send Command:

**Door 1**

1. 0007D082 -> Bell to Relay Door 1
2. 1 Second Delay
3. 16CFEA82

**Door 2**

1. 0003E882 -> Bell to Relay Door 1
2. 1 Second Delay
3. 16CFEA82

lukaiser, 2025-05-18:
I was able to solve it with a HA Automation and a helper button.
Thanks for looking into it!

closed 2025-05-18 by lukaiser