AzonInc/Doorman #48: Muting the apartment station/door bell

Nerdiyde, 2025-06-07:
### Description

Hi guys,
not 100% a feature request (yet):

Does anybody know if it is possible to use doorman to mute the station/doorbell within the apartment. 
So far I could not observe that there is a bus command sent when the mute button on the panel is pressed.
Probably there is a GPIO on the panel that could be used? Does anybody know something about that?

The idea is that I could automatically mute the bell in case my smart home is in a certain mode (sleep, etc).

Thanks in advance and best regards
Fab

AzonInc, 2025-06-07:
Theres no easy way to achieve it.
Someone is using the Doorman relay to disconnect the speaker of the indoor station to be able to mute it.

While you can read/write the mute state, the indoor station won't accept the new setting.
Maybe we could just send a bunch of dummy commands to supress the ringtone.

Nerdiyde, 2025-06-07:
Thanks for that, I will try it.
Can you give me some more info about the last sentence ("Combining this with the settings (ringtone volume) it should be better than nothing"):
Is there an official way to set the ringtone volume via bus commands?

Nerdiyde, 2025-06-08:
Uhhh, I missed that, thanks. 
I will check it and come back with feedback. :)

AzonInc, 2025-06-08:
I think the old interrupt based implementation in master branch would be able to prevent the indoor station from even receiving the command at all by interrupting the transmission.

Shorting the bus for 7ms before the last bit is received (when it's a doorbell command for your serial number)

Nerdiyde, 2025-06-24:
Thanks again for all the hints. I installed the latest version and checked the Volume option. Unfortunately it's not possible to lower the volume to zero (at least then the volume is not off with the lowest setting). 
My solution for now is to deactivate the speaker permanently at the indoor panel and then react only via doorman events on the doorbell. 
(Unfortunately there seems to be a restart issue with the current release, but I will open another issue for that.)
At that point thanks again for the awesome work on that project. :)
Best regards
Fab