AzonInc/Doorman #49: Different behavior depending on the power source

MHordecki, 2025-06-08:
Hi!

I've been setting up my Doorman unit recently. There is not enough current on the intercom bus so the unit is powered from the USB-C port.

In my custom YAML config, I've set up the unit to open the door after the entrance doorbell is pressed four times in quick succession. This works nicely, but only when the Doorman unit is powered from a Macbook (I have two Macbooks at home, both work). When I power the Doorman from a wall USB-C charger or a USB-C powerbank, the button is not recognized at all. Other functions, such as the web interface or MQTT-driven control of the TC bus, are unaffected.

This is very bizarre. I've attached log files with level=VERBOSE, collected through MQTT. Nothing there stands out to me. When running off wall socket power, logs for binary_sensor simply do not appear.

Thanks in advance for any ideas!

[doorman-s3.yaml.txt](https://github.com/user-attachments/files/20646308/doorman-s3.yaml.txt)
[mqtt-log-with-wallpower.txt](https://github.com/user-attachments/files/20646310/mqtt-log-with-wallpower.txt)
[mqtt-log-with-macbook.txt](https://github.com/user-attachments/files/20646309/mqtt-log-with-macbook.txt)

AzonInc, 2025-06-08:
That sounds like a ground noise issue.

The same issue might occur when you connect your MacBook to the charger while it's connected to the Doorman PCB. I tested powering the PCB with my notebook, which worked fine—until I charged it.

You could try using an isolated power supply. Some power supplies work, while others don't, as they seem to cause a significant amount of noise.

I'm going to verify the isolator solution in 1–2 weeks when I can test it.

Let me know if you find out anything new.

Might be worth adding possible solutions to the docs as well.

AzonInc, 2025-06-08:
I'll also verify which hardware revisions are affected, since I have one of each version here.

Hopefully, we can identify whether the issue can be fixed by changing the design.

MHordecki, 2025-06-09:
Good observation! I've repeated the test using a plugged-in Macbook and indeed it doesn't recognize the button press anymore. I've also repeated the test with an USB-C power bank. I had previously reported that this combination also doesn't work; upon retry, I can report that it does in fact work, it's just that the power bank turns off due to the very low power draw of the ESP32. Last time, I had been too slow in pressing the buttons and the power bank had powered down the device.

I've also tried the experiment with my bench-top power supply - it didn't work, but the power supply isn't fully isolated, so it makes sense.

I'm considering getting an isolated AC/DC power supply, as you have suggested. Would something like [Traco Power TMPW](https://www.tracopower.com/series/tmpw-5-t) fit the bill?

AzonInc, 2025-06-17:
I tried a one of the USB-Isolator based on Adum3160, which workfs fine for now using my computer.
For example [this one](https://www.amazon.de/Tsadeer-Isolator-ADUM3160-Digital-Unterst%C3%BCTzt-Transparentes-Schwarz/dp/B0BV11F63M/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2O50TNXYBSRHX&dib=eyJ2IjoiMSJ9.68WNtE6rnZmlECLRYWPJgbiE7Puuh79zJ0vBZDjvvpLzcSIaXRNiHquTPtYPD69smsuUNUhFFQCnDu-cQuTaFIgF6P1Iq1EE7U5HvDok8rziuwe6p4lJmFzJbAu1hONdKg9HDBd7d_r2laHdlUe0R2C9UMFMQp8iLQyZ67fk8l2saX4IU81TIMgfvk0qAJNMNbkWP4MjgXltY9efLieIIuGiTh5grcwYV-0Jwn4Mfm8.44jkjJIt0rLmkaKsjME0yPMEXbz38M62Ngyf_vBdBx0&dib_tag=se&keywords=USB-Isolator%2FUSB-zu-USB-Isolation+Digitales+Signal+Audio-Leistungs+isolator+adum3160&qid=1750148279&sprefix=usb-isolator%2Fusb-zu-usb-isolation+digitales+signal+audio-leistungs+isolator+adum3160%2Caps%2C102&sr=8-6)

But I got mine on aliexpress for ~ 4€

AzonInc, 2025-10-16:
I've updated the documentation to include a note about the issue and a potential solution, so others are aware of it.